oe-alliance/oe-alliance-core #819: SD card mount all partitions to mmc

ArnoldSchiller, 2023-03-17:
I was annoyed as soon as mdev mmc detects that all partitions are mounted on top of each other to /media/mmc. So partition 1 to /media/mmc and then partition 2 to /media/mmc. So only the last partition in /media/mmc is visible. With the change mount-helper.sh should continue to mount to mmc without partition, but if partitions exist the first partition /media/mmcblk and the second partition /media/mmcblk1p2 and so on. 

--- mount-helper.sh
+++ /etc/udev/mount-helper.sh
@@ -190,9 +190,9 @@
 				elif [ "$MODEL1" == "SD	            " ]; then
 					DEVICETYPE="mmc"
 				elif [ "$MODEL1" == "SD              " ]; then
-					DEVICETYPE="mmc"
+					DEVICETYPE="mmcblk"
 				elif [ "$MODEL1" == "SD" ]; then
-					DEVICETYPE="mmc"
+					DEVICETYPE="mmcblk"
 				else
 					#echo "[mdev-mount.sh] USB device found" >> $LOG
 					if [ $PARTNUM -eq "1" -o $PARTNUM -eq "5" ] ; then

Huevos, 2023-03-17:
What image are you using?

ArnoldSchiller, 2023-03-17:
openatv 7.2

ArnoldSchiller, 2023-03-17:
Now I highly doubt this is a receiver story - the kernel is Linux maxytecmultise 4.4.35 #1 SMP Tue Feb 18 23:44:49 UTC 2020 armv7l GNU/Linux - but then again the script doesn't do a label and hardcode hdd and usb. Instead of sda1 or hdd1 or media/$LABEL

TwolDE2, 2023-03-17:
Well rightly or wrongly, images  have this path hard coded (media/mmc), so this change would impact current code in most OE-A images. So unless you are going to go around and change them all, don,t see this code change happening

closed 2026-10-09 by atvcaptain