oe-alliance/oe-alliance-core #824: Error

elhawaryna, 2023-05-14:
root@dm820:~# ENIGMA_DEBUG_LVL=4 enigma2
19:09:17.8105 [Enigma] Python path is '/usr/lib/enigma2/python'.
19:09:17.8110 [Enigma] DVB API version 5, DVB API version minor 11.
19:09:17.8111 [Enigma] Enigma debug level 4.
19:09:17.9849 [Avahi] avahi_watch_new(3 0x1)
19:09:17.9897 [Avahi] client state: 2
19:09:17.9914 [eInit] + (1) Background File Eraser
19:09:17.9918 [eInit] + (5) Tuxtxt
19:09:17.9919 [eInit] + (8) graphics acceleration manager
19:09:17.9928 [eInit] + (9) Font Render Class
19:09:17.9930 [Font] Initializing lib.
19:09:17.9934 [Font] Loading fonts.
19:09:17.9936 [Font] Intializing font cache, using max. 4MB.
19:09:17.9939 [eInit] + (9) gLCD
19:09:17.9940 [eFbLCD] /dev/fb1: Function not implemented
19:09:17.9942 [eFbLCD] framebuffer /dev/fb1 not available
19:09:17.9945 [eLCD] found OLED display!
19:09:17.9950 [eDboxLCD] xres=96, yres=64, bpp=16 lcd_type=3
19:09:17.9952 [eLCD] (96x64x16) buffer 0xbf2988 12288 bytes, stride 192, boxtype: dm820

19:09:17.9954 [gLCDDC] resolution: 96x64x16 stride=192
19:09:17.9955 [eInit] + (9) GFBDC
19:09:17.9957 [fb] /dev/fb0: 7200k video mem
19:09:17.9959 [fb] Using ION allocator
19:09:17.9964 [fb] 32768kB available for acceleration surfaces (via ION).
19:09:18.0160 [fb] double buffering available!
19:09:18.0162 [fb] 2 page(s) available!
19:09:18.1549 [gFBDC] resolution: 1280 x 720 x 32 (stride: 5120) pages: 2
19:09:18.1552 [eInit] + (10) gRC
19:09:18.1557 [gRC] Thread created successfully.
19:09:18.1559 [eInit] + (15) eWindowStyleManager
19:09:18.1560 [eInit] + (20) AVSwitch Driver
19:09:18.1562 [eAVSwitch] failed to open /dev/dbox/fp0 to monitor vcr scart slow blanking changed: Function not implemented
19:09:18.1563 [eInit] + (20) misc options
19:09:18.1565 [eInit] + (20) RC Input layer
19:09:18.1566 [eInit] + (20) UHF Modulator
19:09:18.1568 [eRFmod] couldnt open /dev/rfmod0: Function not implemented
19:09:18.1569 [eInit] + (20) DVB-CI UI
19:09:18.1570 [eInit] + (21) input device driver
19:09:18.1573 [eInputDeviceInit] adding device /dev/input/event0
19:09:18.1575 [eRCInputEventDriver] devicename=dreambox front panel
19:09:18.1577 [eRCInputEventDriver] devicename=dreambox front panel
19:09:18.1578 [eRCDeviceInputDev] device "dreambox front panel" is a remotecontrol
19:09:18.1580 [eInputDeviceInit] adding device /dev/input/event1
19:09:18.1582 [eRCDeviceInputDev] device "dreambox advanced remote control (native)" is a remotecontrol
19:09:18.1584 [eInputDeviceInit] adding device /dev/input/event2
19:09:18.1586 [eRCInputEventDriver] devicename=dreambox advanced remote control (native)
19:09:18.1587 [eRCInputEventDriver] devicename=dreambox advanced remote control (native)
19:09:18.1588 [eRCDeviceInputDev] device "dreambox advanced remote control (native)" is a remotecontrol
19:09:18.1590 [eInputDeviceInit] adding device /dev/input/event3
19:09:18.1591 [eRCInputEventDriver] devicename=dreambox ir keyboard
19:09:18.1592 [eRCInputEventDriver] devicename=dreambox ir keyboard
19:09:18.1594 [eRCDeviceInputDev] device "dreambox ir keyboard" is a keyboard
19:09:18.1595 [eInputDeviceInit] adding device /dev/input/event4
19:09:18.1597 [eRCInputEventDriver] devicename=dreambox ir mouse
19:09:18.1598 [eRCInputEventDriver] devicename=dreambox ir mouse
19:09:18.1599 [eRCDeviceInputDev] device "dreambox ir mouse" is a mouse
19:09:18.1601 [eInputDeviceInit] adding device /dev/input/event5
19:09:18.1602 [eRCInputEventDriver] devicename=dreambox remote control (bluetooth le)
19:09:18.1604 [eRCInputEventDriver] devicename=dreambox remote control (bluetooth le)
19:09:18.1605 [eRCDeviceInputDev] device "dreambox remote control (bluetooth le)" is a remotecontrol
19:09:18.1607 [eInputDeviceInit] Found 6 input devices.
19:09:18.1608 [eInit] + (22) Hdmi CEC driver
19:09:18.1629 [eHdmiCEC] detected physical address change: 1000 --> 3000
19:09:18.1630 [eHdmiCEC] send message 0F 84 30 00 03
19:09:18.1638 [eInit] + (30) eActionMap
19:09:18.1639 [eInit] + (35) CA handler
19:09:18.1643 [eInit] + (35) CI Slots
19:09:18.1645 [CI] scanning for common interfaces..
19:09:18.1649 [CI] done, found 0 common interface slots
[... 205 lines cut ...]
19:09:27.2118 [eInit] - (41) eServiceFactoryDVD
19:09:27.2119 [eInit] - (40) eServiceCenter
19:09:27.2119 [eServiceCenter] clear instance
19:09:27.2120 [eInit] - (35) CA handler
19:09:27.2122 [eInit] - (35) CI Slots
19:09:27.2122 [eInit] - (30) eActionMap
19:09:27.2123 [eInit] - (22) Hdmi CEC driver
19:09:27.2124 [eInit] - (21) input device driver
19:09:27.3166 [eInit] - (20) AVSwitch Driver
19:09:27.3168 [eInit] - (20) misc options
19:09:27.3169 [eInit] - (20) RC Input layer
19:09:27.3170 [eInit] - (20) UHF Modulator
19:09:27.3171 [eInit] - (20) DVB-CI UI
19:09:27.3173 [eInit] - (15) eWindowStyleManager
19:09:27.3174 [eInit] - (10) gRC
19:09:27.3176 [gRC] Waiting for gRC thread shutdown.
19:09:27.3183 [gRC] Thread has finished.
19:09:27.3185 [eInit] - (9) Font Render Class
19:09:27.3187 [eInit] - (9) gLCD
19:09:27.3188 [eInit] - (9) GFBDC
19:09:27.3193 [eInit] - (8) graphics acceleration manager
19:09:27.3194 [eInit] - (5) Tuxtxt
19:09:27.3196 [eInit] - (1) Background File Eraser
19:09:27.3197 [eInit] reached rl -1
root@dm820:~#

atvcaptain, 2023-05-15:
i have install latest openatv 7.3 dm820 and start no crash

jbleyel, 2023-05-15:
Please provide this file /usr/lib/enigma2/python/StartEnigma.py

atvcaptain, 2023-05-16:
dm820 works here without any issue, we close the issue report

closed 2023-05-16 by atvcaptain