oe-alliance/oe-alliance-core #839: This (uchardet) does not work, please fix it

atefganm, 2023-08-27:
`atef@atef-ganm:~/build-enviroment-5.3/build-enviroment$ MACHINE=dm820 DISTRO=opendroid DISTRO_TYPE=release make image
Generating /home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/env.source
Generating /home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/conf/opendroid.conf
Generating /home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/conf/bblayers.conf
Generating /home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/conf/local.conf
check internet connection: Online      
NOTE: Your conf/bblayers.conf has been automatically updated.
NOTE: Your conf/bblayers.conf has been automatically updated.
NOTE: Your conf/bblayers.conf has been automatically updated.
Loading cache: 100% |                                                                                                                                                                                                                         | ETA:  --:--:--
Loaded 0 entries from dependency cache.
ERROR: ExpansionError during parsing /home/atef/build-enviroment-5.3/build-enviroment/meta-oe-alliance/meta-oe/recipes-support/uchardet/uchardet_git.bb#############################################################                           | ETA:  0:00:15
Traceback (most recent call last):
  File "/home/atef/build-enviroment-5.3/build-enviroment/bitbake/lib/bb/fetch2/__init__.py", line 1221, in srcrev_internal_helper(ud=<bb.fetch2.FetchData object at 0x7fc0d12f7370>, d=<bb.data_smart.DataSmart object at 0x7fc0d11d0130>, name='default'):
             d.setVar("__BBAUTOREV_ACTED_UPON", True)
    >        srcrev = ud.method.latest_revision(ud, d, name)
     
  File "/home/atef/build-enviroment-5.3/build-enviroment/bitbake/lib/bb/fetch2/__init__.py", line 1631, in Git.latest_revision(ud=<bb.fetch2.FetchData object at 0x7fc0d12f7370>, d=<bb.data_smart.DataSmart object at 0x7fc0d11d0130>, name='default'):
             except KeyError:
    >            revs[key] = rev = self._latest_revision(ud, d, name)
                 return rev
  File "/home/atef/build-enviroment-5.3/build-enviroment/bitbake/lib/bb/fetch2/git.py", line 746, in Git._latest_revision(ud=<bb.fetch2.FetchData object at 0x7fc0d12f7370>, d=<bb.data_smart.DataSmart object at 0x7fc0d11d0130>, name='default'):
     
    >        output = self._lsremote(ud, d, "")
             # Tags of the form ^{} may not work, need to fallback to other form
  File "/home/atef/build-enviroment-5.3/build-enviroment/bitbake/lib/bb/fetch2/git.py", line 729, in Git._lsremote(ud=<bb.fetch2.FetchData object at 0x7fc0d12f7370>, d=<bb.data_smart.DataSmart object at 0x7fc0d11d0130>, search=''):
                     bb.fetch2.check_network_access(d, cmd, repourl)
    >            output = runfetchcmd(cmd, d, True)
                 if not output:
  File "/home/atef/build-enviroment-5.3/build-enviroment/bitbake/lib/bb/fetch2/__init__.py", line 934, in runfetchcmd(cmd='export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="2911"; export SSH_AUTH_SOCK="/tmp/ssh-XXXXXXqsHZpw/agent.2910"; export PATH="/home/atef/build-enviroment-5.3/build-enviroment/openembedded-core/scripts:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/bin/mipsel-oe-linux:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot/usr/bin/crossscripts:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/sbin:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/bin:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/sbin:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/bin:/home/atef/build-enviroment-5.3/build-enviroment/bitbake/bin:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/hosttools"; export HOME="/home/atef"; git -c gc.autoDetach=false -c core.pager=cat ls-remote https://gitlab.freedesktop.org/uchardet/uchardet.git ', d=<bb.data_smart.DataSmart object at 0x7fc0d04f9180>, quiet=True, cleanup=[], log=None, workdir=None):
     
    >        raise FetchError(error_message)
     
bb.data_smart.ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception FetchError: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="2911"; export SSH_AUTH_SOCK="/tmp/ssh-XXXXXXqsHZpw/agent.2910"; export PATH="/home/atef/build-enviroment-5.3/build-enviroment/openembedded-core/scripts:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/bin/mipsel-oe-linux:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot/usr/bin/crossscripts:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/sbin:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/bin:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/sbin:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/work/mips32el-oe-linux/uchardet/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/bin:/home/atef/build-enviroment-5.3/build-enviroment/bitbake/bin:/home/atef/build-enviroment-5.3/build-enviroment/builds/opendroid/release/dm820/tmp/hosttools"; export HOME="/home/atef"; git -c gc.autoDetach=false -c core.pager=cat ls-remote https://gitlab.freedesktop.org/uchardet/uchardet.git  failed with exit code 128, output:
fatal: unable to access 'https://gitlab.freedesktop.org/uchardet/uchardet.git/': The requested URL returned error: 504

The variable dependency chain for the failure is: SRCPV -> PV -> BP -> FILESPATH

ERROR: Parsing halted due to errors, see error messages above

Summary: There were 2 ERROR messages, returning a non-zero exit code.
make: *** [Makefile:1031: image] Error 1
atef@atef-ganm:~/build-enviroment-5.3/build-enviroment$ 
`

atefganm, 2023-08-27:
The problem has been resolved with this link

SRC_URI = "git://github.com/BYVoid/uchardet.git;protocol=https;branch=master"

atvcaptain, 2023-08-28:
https://github.com/oe-alliance/oe-alliance-core/commit/8b4b35e29417a5af7cdbfb4272f421e327c39c2b

closed 2023-08-28 by atvcaptain