oe-alliance/oe-alliance-core #843: wake on lan on SF8008

bibcoder, 2023-11-14:
Hey there,

the Octagon SF8008 uses a Hi3798MV200 chipset, which supports according to
https://www.hisilicon.com/en/products/smart-media/STB/Hi3798MV200

"One 1000 Mbit/s Ethernet port or one 100 Mbit/s Ethernet port, one GMAC and one FE PHY integrated"

Thus, to me this seems, it supports Wake on lan by the hardware chipset -- I may be wrong. On the other hand, no matter which linux distro I use, this device still uses 6 Watts in Standby, when WOL is activated. Could it be, that the software side doesn't yet support the full features of the low power mode of this chipset?

kind regards

atvcaptain, 2023-11-14:
we use it as the manufacturer has integrated it into the driver, oe-a has no influence on it.

bibcoder, 2023-11-15:
Good Morning!

Thanks for your reply. So with "manufacturer" you mean the producer of the chipset, right? You mean that the oe-a code just calls the according Wake-on-lan and power mode functions provided by the driver for the chipset, and this is kind of a black box what happens next, did I understand you correct?

have a nice day

atvcaptain, 2023-11-15:
yes the manufacturer of the box and the driver, and yes for us it is a black box, we have no possibility to intervene.

closed 2023-11-15 by atvcaptain

bibcoder, 2023-11-15:
Thank you atvcaptain. I will try to find the root of this evil, the manufacturer ;-)