oe-alliance/oe-alliance-core #872: Please update the patch for ZeroTier

atefganm, 2024-05-03:
Sstate summary: Wanted 1880 Local 1841 Mirrors 0 Missed 39 Current 4337 (97% match, 99% complete)
Removing 12 stale sstate objects for arch dm820: 100% |########################################################################################################################################################################################| Time: 0:00:00
NOTE: Executing Tasks
ERROR: zerotier-1.12.0+gitAUTOINC+91e7ce87f0-r0 do_patch: Applying patch '0001-accept-external-ldflags.patch' on target directory '/home/atef/build-enviroment-5.3/build-enviroment/builds/teamblue/release/dm820/tmp/work/mips32el-oe-linux/zerotier/1.12.0+gitAUTOINC+91e7ce87f0-r0/git'
CmdError('quilt --quiltrc /home/atef/build-enviroment-5.3/build-enviroment/builds/teamblue/release/dm820/tmp/work/mips32el-oe-linux/zerotier/1.12.0+gitAUTOINC+91e7ce87f0-r0/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0001-accept-external-ldflags.patch
patching file make-linux.mk
Hunk #1 succeeded at 26 (offset 1 line).
Hunk #2 succeeded at 35 (offset 1 line).
Hunk #3 FAILED at 70.
Hunk #4 succeeded at 333 (offset 6 lines).
Hunk #5 succeeded at 360 (offset 6 lines).
1 out of 5 hunks FAILED -- rejects in file make-linux.mk
Patch 0001-accept-external-ldflags.patch does not apply (enforce with -f)

stderr: ')
ERROR: Logfile of failure stored in: /home/atef/build-enviroment-5.3/build-enviroment/builds/teamblue/release/dm820/tmp/work/mips32el-oe-linux/zerotier/1.12.0+gitAUTOINC+91e7ce87f0-r0/temp/log.do_patch.180209
ERROR: Task (/home/atef/build-enviroment-5.3/build-enviroment/meta-oe-alliance/meta-oe/recipes-connectivity/zerotier/zerotier_1.12.0.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 12743 tasks of which 12670 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/atef/build-enviroment-5.3/build-enviroment/meta-oe-alliance/meta-oe/recipes-connectivity/zerotier/zerotier_1.12.0.bb:do_patch
Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.
make: *** [Makefile:1031: image] Error 1
atef@atef-ganm:~/build-enviroment-5.3/build-enviroment$

atefganm, 2024-05-03:
The patch has been updated
[0001-accept-external-ldflags.patch.zip](https://github.com/oe-alliance/oe-alliance-core/files/15197762/0001-accept-external-ldflags.patch.zip)

nickersk, 2024-05-03:
you have to wait, test builds are running already

nickersk, 2024-05-03:
fixed here: https://github.com/oe-alliance/oe-alliance-core/commit/806fea7b82c2c60d4a3695c75a9cf33b92c83c51