oe-alliance/oe-alliance-core #882: Unable to get checksum for enigma2-plugins SRC_URI entry fix-missing-function-declaration.patch: file could not be found

atefganm, 2024-07-11:
atef@atef-ganm:~/build-enviroment-5.5/build-enviroment$ MACHINE=dm820 DISTRO=openvix DISTRO_TYPE=release make image
Generating /home/atef/build-enviroment-5.5/build-enviroment/builds/openvix/release/dm820/env.source
Generating /home/atef/build-enviroment-5.5/build-enviroment/builds/openvix/release/dm820/conf/openvix.conf
Generating /home/atef/build-enviroment-5.5/build-enviroment/builds/openvix/release/dm820/conf/bblayers.conf
Generating /home/atef/build-enviroment-5.5/build-enviroment/builds/openvix/release/dm820/conf/local.conf
check internet connection: Online      
NOTE: Your conf/bblayers.conf has been automatically updated.
NOTE: Your conf/bblayers.conf has been automatically updated.
NOTE: Your conf/bblayers.conf has been automatically updated.
Loading cache: 100% |                                                                                                                                                                                                                         | ETA:  --:--:--
Loaded 0 entries from dependency cache.
ERROR: /home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins.bb: Unable to get checksum for enigma2-plugins SRC_URI entry fix-missing-function-declaration.patch: file could not be found
The following paths were searched:
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/openvix/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/openvix/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/openvix/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/dreambox/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/dreambox/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/dreambox/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/dm820/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/dm820/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/dm820/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/bcm7435/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/bcm7435/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/bcm7435/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/mips32el/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/mips32el/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/mips32el/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/dm820/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/dm820/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/dm820/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/mipsarcho32el/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/mipsarcho32el/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/mipsarcho32el/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/mipsarcho32/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/mipsarcho32/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/mipsarcho32/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/mipsarchel/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/mipsarchel/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/mipsarchel/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/mipsarch/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/mipsarch/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/mipsarch/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/mipsel/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/mipsel/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/mipsel/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins-6.6+gitr/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/enigma2-plugins/fix-missing-function-declaration.patch
/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/enigma2-plugins/files/fix-missing-function-declaration.patch
ERROR: Parsing halted due to errors, see error messages above

Summary: There were 2 ERROR messages, returning a non-zero exit code.
make: *** [Makefile:1031: image] Error 1
atef@atef-ganm:~/build-enviroment-5.5/build-enviroment$

atvcaptain, 2024-07-11:
vix and bh not start porting to 5.5 you must wait some month, we have start now check all for openatv this is not ready too

ahmedmoselhi, 2024-08-14:
same issue here

closed 2024-12-14 by atefganm