oe-alliance/oe-alliance-core #889: The box enters in the bootmenu instead loads a system from a slot.

stachu99, 2024-08-20:
Occurs when the box is in deep standby mode and turns on by short pressing the red button on the remote control.
The same situation with power timers, those timers wake up the box and stays in the bootmenu.
I need manually choose a slot to load a sytem.
Pressing the rear power button or restart an active system loads last loaded system correctly.

Box: Zgemma H9S SE
Recovery_emmc installed from OpenATV 9 version date 14th or 15th Aug 2024.
All four slots removed directly in the bootmenu, Android system set factory reset.
Installed OpenATV 9 version from 10.08.2014 at slot 1.
After all at /dev/mmcblk0p23 is one directory only, linuxrootfs1.
/dev/mmcblk0p4,  STARTUP and STARTUP_LINUX_1:
`bootcmd=setenv notee y; setenv bootargs $(bootargs) $(bootargs_common); mmc read 0 0x1000000 0x3BD000 0x8000; bootm 0x1000000; run bootcmd_fallback
bootargs=root=/dev/mmcblk0p23 rootsubdir=linuxrootfs1 rootfstype=ext4 kernel=/dev/mmcblk0p19`

I tested with installed OpenATV, OpenPLI and OpenVix also flashed recovery_emmc of each distribution for the test purpose.

I attached files with debug logs. The file BAD_DeepStandby_RemoteControlShortPressing.txt contains the logs, the box enters in the bootmenu instead loads the system at the slot 1.

[BAD_DeepStandby_RemoteControlShortPressing.txt](https://github.com/user-attachments/files/16681459/BAD_DeepStandby_RemoteControlShortPressing.txt)
[OK_DeepStandby_RemoteControlLongPressing.txt](https://github.com/user-attachments/files/16681460/OK_DeepStandby_RemoteControlLongPressing.txt)
[OK_RearButton.txt](https://github.com/user-attachments/files/16681461/OK_RearButton.txt)
[OK_RearButtonAndRemoteControlLongPressing.txt](https://github.com/user-attachments/files/16681462/OK_RearButtonAndRemoteControlLongPressing.txt)
[OK_Restart_system_Slot1.txt](https://github.com/user-attachments/files/16681463/OK_Restart_system_Slot1.txt)

Hopely you know solution.
Best regards, Stachu99

AbuBaniaz, 2026-10-12:
This is not fixed yet.

Put receiver into deep standby, restart, it will boot at bootloader menu.

atvcaptain, 2026-10-16:
Thank you for confirming that the issue still occurs.
This behaviour is related to the receiver’s fastboot/bootloader implementation. We cannot change that behaviour from Enigma2, and the manufacturer no longer provides the support needed to address it.
We therefore have no fix available and are closing this ticket as a known, unresolved limitation—not as fixed.
Starting via the rear power switch or restarting from a running image works according to the original report. Automatic wake-up from deep standby remains affected and should not be relied upon for unattended operation.

closed 2026-10-16 by atvcaptain