oe-alliance/oe-alliance-core #894: ERROR: oe-alliance-feeds-6.6-r20 do_package_qa: QA Issue: oe-alliance-feeds rdepends on enigma2-plugin-extensions-cdinfo, but it isn't a build dependency? [build-deps]

atefganm, 2024-09-13:
`NOTE: Executing Tasks
ERROR: oe-alliance-feeds-6.6-r20 do_package_qa: QA Issue: oe-alliance-feeds rdepends on enigma2-plugin-extensions-cdinfo, but it isn't a build dependency? [build-deps]
ERROR: oe-alliance-feeds-6.6-r20 do_package_qa: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/atef/build-enviroment-5.5/build-enviroment/builds/openvix/release/dm820/tmp/work/dm820-oe-linux/oe-alliance-feeds/oe-alliance-feeds-6.6-r20/temp/log.do_package_qa.913622
ERROR: Task (/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/image/oe-alliance-feeds.bb:do_package_qa) failed with exit code '1'
NOTE: Tasks Summary: Attempted 19625 tasks of which 19499 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/image/oe-alliance-feeds.bb:do_package_qa
    log: /home/atef/build-enviroment-5.5/build-enviroment/builds/openvix/release/dm820/tmp/work/dm820-oe-linux/oe-alliance-feeds/oe-alliance-feeds-6.6-r20/temp/log.do_package_qa.913622
Summary: There were 2 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.
make: *** [Makefile:1034: image] Error 1
atef@atef-ganm:~/build-enviroment-5.5/build-enviroment$ 
`

atefganm, 2024-09-21:
Unfortunately, the problem still exists until now.

`Build Configuration:
BB_VERSION           = "2.9.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "linuxmint-22"
TARGET_SYS           = "mipsel-oe-linux"
MACHINE              = "dm820"
DISTRO               = "openvix"
DISTRO_VERSION       = "6.6"
TUNE_FEATURES        = "o32 fpu-hard mips32"
TARGET_FPU           = ""
meta-local           = "5.5:d7b5acf5d55275c8bb9786f3a85aaafef0864edb"
meta-oe              = "HEAD:dcd9f9e619a5d849bedebbed1518111440e4ff77"
meta                 = "HEAD:b4cf6d5236a3eacaf56ca2f805b006efac65b26c"
meta-oe              
meta-multimedia      
meta-networking      
meta-filesystems     
meta-python          
meta-webserver       = "HEAD:84d6e138cb558cea9155e5a392e2ff7647e2e688"
meta-python2         = "HEAD:c6e056ccbaaadfdcead36fae0a0f466f802e4e66"
meta-qt5.15          = "HEAD:a9df95e23e3864089eb025f70de745b4f2619a62"
meta-clang           = "HEAD:814d20901a71fd4d98a5a67789d69ef7e9a2169c"
meta-abcom           
meta-airdigital      
meta-amiko           
meta-anadol          
meta-ax              
meta-blackbox        
meta-beyonwiz        
meta-ceryon          
meta-dags            
meta-dinobot         
meta-dream           
meta-edision         
meta-entwopia        
meta-formuler        
meta-gfutures        
meta-gigablue        
meta-ini             
meta-maxytec         
meta-broadmedia      
meta-odin            
meta-octagon         
meta-protek          
meta-qviart          
meta-skylake         
meta-tiviar          
meta-tripledot       
meta-uclan           
meta-ultramini       
meta-vuplus          
meta-xp              
meta-xtrend          
meta-xcore           = "HEAD:dcd9f9e619a5d849bedebbed1518111440e4ff77"

Sstate summary: Wanted 1106 Local 880 Mirrors 0 Missed 226 Current 8841 (79% match, 97% complete)###################################################################################################                    | ETA:  0:00:01
Removing 28 stale sstate objects for arch allarch: 100% |###############################################################################################################################################################| Time: 0:00:00
Removing 104 stale sstate objects for arch dm820: 100% |################################################################################################################################################################| Time: 0:00:03
Removing 75 stale sstate objects for arch mips32el: 100% |##############################################################################################################################################################| Time: 0:00:03
NOTE: Executing Tasks
ERROR: oe-alliance-feeds-6.6-r20 do_package_qa: QA Issue: oe-alliance-feeds rdepends on enigma2-plugin-extensions-cdinfo, but it isn't a build dependency? [build-deps]
ERROR: oe-alliance-feeds-6.6-r20 do_package_qa: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/atef/build-enviroment-5.5/build-enviroment/builds/openvix/release/dm820/tmp/work/dm820-oe-linux/oe-alliance-feeds/oe-alliance-feeds-6.6-r20/temp/log.do_package_qa.143152
ERROR: Task (/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/image/oe-alliance-feeds.bb:do_package_qa) failed with exit code '1'
NOTE: Tasks Summary: Attempted 19761 tasks of which 19337 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-oe-alliance/image/oe-alliance-feeds.bb:do_package_qa
    log: /home/atef/build-enviroment-5.5/build-enviroment/builds/openvix/release/dm820/tmp/work/dm820-oe-linux/oe-alliance-feeds/oe-alliance-feeds-6.6-r20/temp/log.do_package_qa.143152
Summary: There was 1 WARNING message.
Summary: There were 2 ERROR messages, returning a non-zero exit code.
make: *** [Makefile:1034: image] Error 1
atef@ganm:~/build-enviroment-5.5/build-enviroment$ 
`

TwolDE2, 2024-09-21:
Remove dvd from machine features:
https://github.com/oe-alliance/oe-alliance-core/blob/5.4/meta-brands/meta-dream/conf/machine/dm820.conf

closed 2024-12-14 by atefganm