oe-alliance/oe-alliance-core #896: openbh>>blackhole-base.bb:do_install  ERROR

atefganm, 2024-10-01:
ERROR: blackhole-base-1.0-r21 do_install: Execution of '/home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/temp/run.do_install.291529' failed with exit code 1
ERROR: Logfile of failure stored in: /home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/temp/log.do_install.291529
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/atef/build-enviroment-5.5/build-enviroment/openembedded-core/meta/recipes-core/glibc/glibc_2.40.bb:do_populate_sysroot', '/home/atef/build-enviroment-5.5/build-enviroment/openembedded-core/meta/recipes-devtools/gcc/gcc-cross_14.2.bb:do_populate_sysroot', '/home/atef/build-enviroment-5.5/build-enviroment/openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_14.2.bb:do_populate_sysroot', '/home/atef/build-enviroment-5.5/build-enviroment/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.68.bb:do_populate_sysroot', 'virtual:native:/home/atef/build-enviroment-5.5/build-enviroment/openembedded-core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/atef/build-enviroment-5.5/build-enviroment/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['linux-libc-headers', 'gettext-minimal-native', 'glibc', 'binutils-cross-mipsel', 'gcc-cross-mipsel', 'gcc-runtime', 'libgcc', 'libtool-native', 'm4-native', 'quilt-native', 'texinfo-dummy-native', 'zlib-native', 'flex-native', 'gnu-config-native', 'patch-native', 'pseudo-native', 'xz-native', 'zstd-native', 'attr-native', 'gmp-native', 'libmpc-native', 'mpfr-native', 'sqlite3-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| install: cannot stat '/home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/sources/bhmachine': No such file or directory
| WARNING: /home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/temp/run.do_install.291529:150 exit 1 from 'install -m 0644 /home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/sources/bhmachine /home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/image/etc/bhmachine'
| WARNING: Backtrace (BB generated script):
| 	#1: do_install, /home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/temp/run.do_install.291529, line 150
| 	#2: main, /home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/temp/run.do_install.291529, line 187
ERROR: Task (/home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-distros/openbh/blackhole/blackhole-base.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 13265 tasks of which 13080 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/atef/build-enviroment-5.5/build-enviroment/meta-oe-alliance/meta-oe/recipes-distros/openbh/blackhole/blackhole-base.bb:do_install
    log: /home/atef/build-enviroment-5.5/build-enviroment/builds/openbh/release/dm820/tmp/work/mips32el-oe-linux/blackhole-base/blackhole-base-1.0-r21/temp/log.do_install.291529
Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.
make: *** [Makefile:1037: image] Error 1
atef@ganm:~/build-enviroment-5.5/build-enviroment$

Ev0-BH, 2024-10-01:
openbh does not support dreamboxes

atefganm, 2024-10-01:
I used some patches from openAtv to get it working on my device.

But the image is built naturally on branch 5.4.

[image]

atefganm, 2024-10-01:
The problem has been solved.

https://github.com/oe-alliance/oe-alliance-core/pull/897/commits/02c3bf0c540888540b626210d2bf2d943b0633c3

[image]

closed 2024-10-01 by atefganm