oe-alliance/oe-alliance-core #899: vusolo2 build error

WanWizard, 2024-10-26:
````
ERROR: vuplus-dvb-modules-vusolo2-3.13.5+20190429-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'https://source.mynonpublic.com/vuplus/release/vuplus-dvb-modules/vuplus-dvb-modules-vusolo2-3.13.5-20190429.tar.gz'))
````

atvcaptain, 2024-10-26:
we build very long vuplus-dvb-proxy-vusolo2-3.13.5-20190429.r0.tar.gz

i dont have the non proxy driver, if you have i can upload too

WanWizard, 2024-10-26:
So all these recipies in https://github.com/oe-alliance/oe-alliance-core/tree/7edff5d4b83983d980efc05aecc34214794fc115/meta-brands/meta-vuplus/recipes-drivers are no longer used?

atvcaptain, 2024-10-26:
solo2 build proxy and the old not use yes

WanWizard, 2024-10-26:
Ok, I'lve dig it out of the source cache, thanks.

closed 2024-10-26 by WanWizard