oe-alliance/oe-alliance-core #902: KODI 22 NO SOUND AND NO IMAGE IN 4K

ruipisco, 2024-12-09:
Using OPEN ATV 7.5

Describe the bug / Actual behaviour:
After the changes made in 1 of December of 2024, for supporting KODI Standalone, after an fresh install of image and KODI plug-in, the movie playback doesn't have any sound and it cannot play pictures with 4K resolutions.
It does not work with an external E2Player since after adding the rules in KODI playercorefactory.xml it not play in the external player anymore.

Expected behaviour:
Sound working on KODI interface and movies playback.
Image works with all kind of resolutions with or without HDR.

Has this issue started to happen just recently?
It started after the update of OPENATV 7.5 image of 2-12-2024

To reproduce:
Steps to reproduce the behavior:

Install lateste OPENATV;
After that install KODI 22 from plugin browser.
Launch Kodi and you w: using Elementum or other).
In case of launching a movies with 4K resolution (whit or without HDR) there will not be displayed any image (the image only work on max resolution of 1080p).
Image/Box Model (please complete the following information):

Version: OPEN ATV 7.5 (image date:
BoxModel: VU PLUS ZERO 4K

luisteraar, 2024-12-10:
your complaint that some 4k/uhd broadcasts are not displayed has already reached
 me from you via github openatv.
Have already reported that I am solving it.
The source is not yet on oe-alliance it is still in test phase as
also kodi 22-alpha and openatv 7.5 develop.
It is better to report what does work now in kodi on E2 hardware.
Video playback now runs over a new internal kodi sink/renderer so no fallback over external player.
This gives the advantage that all internal kodi features like e.g. teletext/subtitle works.
The cpu load at internal gl/gles sink has been reduced from about 200% to 10 to 15%
Please test further if you find anything else I'm happy to fix it

luisteraar, 2024-12-10:
Just testet for you
kodi works fine an 2160p you have to set the
hdmi resolution in enigma2 config
testet on h9se
let me know if you find something

jbleyel, 2024-12-10:
This is not an enigma issue.

luisteraar, 2024-12-10:
I did some research
4K problem finds its cause in default kodi source
CBitstreamConverter::Open Invalid hvcC
kodi does not support all hvc formats.
It's a matter of reporting the problem to kodi
I am looking for a solution that kodi can handle more hvc formats.
Then oe-alliance kodi is better than standard

luisteraar, 2024-12-11:
4K/UHD playing fixed

ruipisco, 2024-12-12:
@luisteraar I performed a fresh installation of the OPENATV 7.5 image and a fresh setup of the KODI plugin today (12-12-2024). After conducting some tests, I found that streaming playback is still not working for 4K streams.

Additionally, I am using the default sound settings in OPENATV 7.5 and the “default()” option in the sound settings on KODI, but there is still no sound in the KODI interface or during video playback.

luisteraar, 2024-12-12:
post log file
 /tmp/kodi/kodi.log

ruipisco, 2024-12-12:
[kodi.log](https://github.com/user-attachments/files/18109278/kodi.log)

ruipisco, 2024-12-12:
here is my default settings Captures:

1 - KODI Audio Settings:

[image]

2 - OPENATV 7.5 AUDIO SETTINGS DEFAULT

[image]

With alternative OpenATV Audio settings

[image]

No sound with each configuration, hope this will help.

luisteraar, 2024-12-12:
Ahh ok
024-12-12 09:45:40.276 T:4283  warning <general>: ActiveAE - large audio sync error: -9649.024039
2024-12-12 09:45:32.359 T:4959     info <general>: DVDVideoCodecDvb: format:STB-h265
this is 4k and fixed in new release.
reason no audio kodi is not in sync audio with video reason no video in h265
better to discuss here (not over illegal streams)
https://www.opena.tv/viewtopic.php?p=575369#p575369

ruipisco, 2024-12-12:
The sound problem afects also the sound of the user actions in the KODI interface, so the problem is not only because the video is in h265. It's a general problem on the sound on KODI, since it does not work at all. Probably the VuPlus Boxes need some fix to used the native sound driver.

ruipisco, 2024-12-12:
hope this could help: https://forums.openpli.org/topic/80562-all-kodi-related-issues-only-in-openpli-80-and-no-previous-releases/page-10

jbleyel, 2024-12-12:
> The sound problem afects also the sound of the user actions in the KODI interface, so the problem is not only because the video is in h265. It's a general problem on the sound on KODI, since it does not work at all. Probably the VuPlus Boxes need some fix to used the native sound driver.

No way. This is impossible because the source is closed and VU+ will not do any changes for the drivers.
You have to live with it or get a different box.

luisteraar, 2024-12-12:
first you try to display audio where
video and audio are not in sync,it won't work.
then you try to display 2024.2160p.HDR10Plus.DV.WEBRip.6CH.x265.HEVC-PSA without knowing if your hardware can handle it.
alsa wants to do it
Creating audio stream (codec id: 86018, channels: 6, sample rate: 48000, no pass-through)
This is all done in kodi itself for questions about this go to kodi forum.
Several testers on forum openatv have tested it on VU and have no problem.
For me, it can be closed here

ruipisco, 2024-12-12:
OK you can close this issue @luisteraar.
If you say that other users with Vu Plus Zero 4K box with OPENATV 7.5 don't have the issue i will assume that the problem is only in my setup box (but i am 100% sure that it is not the problem). Thank you for all your help.

luisteraar, 2024-12-12:
Ask kodi
mayby enable audio downmix in kodi audio config or
user other audio stream

ruipisco, 2024-12-12:
Already tryed the audiodownmix options in KODI config but it does not solve the issue.

> Just testet for you kodi works fine an 2160p you have to set the hdmi resolution in enigma2 config testet on h9se let me know if you find something

On VuPlus the resolution is 2160p already selected once my TV is a 4K screen.
In KODI the resolution for VuPlus is only at 720p, and it has no option to select 1080p, so the interface is low quality.

Thank you once again.

ruipisco, 2025-03-12:
Kodi 22 alpha 1 on openatv 7.6 is now working properly, transitioning from kodi to external E2Player, with the default sound settings on OpenAtv and default sound configs in kodi.

closed 2025-03-12 by ruipisco

ruipisco, 2025-03-14:
@jbleyel can you please verify the code im my fork, of kodiext is good for KODI to work with the external playser in VU decoders? I have tested it on OPENATV 7.6, and the color space and scaling and colro space of the UI is based on the HDMI type that is conected and it has an updated interface for  the external E2PLayer. Fixed also the Player Screen and Video Info View. Here is the url of my fork: [https://github.com/ruipisco/kodiext/tree/V3](url), if its good you can pull the code and images to the oe-allience/kodiext V3

jbleyel, 2025-03-14:
I have created a new repro.
This repro has now the option to create issues.
So you have read my comments on your commits?

ruipisco, 2025-03-14:
No i am trying to understamd how to work with git hub

jbleyel, 2025-03-14:
Here is the new version.

https://github.com/oe-alliance/kodiext

The new code is in master branch now.