oe-alliance/oe-alliance-core #927: on zgemmash2 missing Module iptable_nat using latest OpenATV 7.5.1

bcarrella, 2025-05-21:
root@zgemmash2:~# modprobe iptable_nat
modprobe: FATAL: Module iptable_nat not found in directory /lib/modules/4.10.12

root@zgemmash2:~# ls /lib/modules/4.10.12/kernel/net/ipv4/netfilter/
ip_tables.ko               ipt_REJECT.ko              iptable_filter.ko          nf_conntrack_ipv4.ko       nf_log_ipv4.ko             nf_nat_masquerade_ipv4.ko
ipt_ECN.ko                 ipt_ah.ko                  iptable_mangle.ko          nf_defrag_ipv4.ko          nf_nat_ipv4.ko             nf_reject_ipv4.ko

bcarrella, 2025-05-24:
> Most likely the NAT module is not built because it has not been enabled in the kernel.
> 
> [oe-alliance-core/meta-brands/meta-airdigital/recipes-linux/linux-airdigital-4.10.12/h3/defconfig](https://github.com/oe-alliance/oe-alliance-core/blob/8c9d0196e303ccb64b87add429438249b5285b52/meta-brands/meta-airdigital/recipes-linux/linux-airdigital-4.10.12/h3/defconfig#L704)
> 
> Line 704 in [8c9d019](/oe-alliance/oe-alliance-core/commit/8c9d0196e303ccb64b87add429438249b5285b52)
> 
>  # CONFIG_NETFILTER_XT_NAT is not set 
> [@atvcaptain](https://github.com/atvcaptain)

@m4dhouse  @atvcaptain  I think that the kernel for zgemmash2 should be the one at
[oe-alliance/oe-alliance-core/blob/8c9d0196e303ccb64b87add429438249b5285b52/meta-brands/meta-airdigital/recipes-linux/linux-airdigital-4.10.12/sh1/defconfig](https://github.com/oe-alliance/oe-alliance-core/blob/8c9d0196e303ccb64b87add429438249b5285b52/meta-brands/meta-airdigital/recipes-linux/linux-airdigital-4.10.12/sh1/defconfig

bcarrella, 2025-05-25:
@WXbet thank you I'll try it as soon as possible. I'll keep you informed.

MastaG, 2025-08-30:
@atvcaptain I know that oe-a derived images are generally meant for watching dvb using enigma2.
But it would be great if all of the defconfigs for all receivers could at least enable the modules required for wireguard and vpn routing:

```
#
# Core crypto framework (must be built-in)
#
CONFIG_CRYPTO=y
CONFIG_CRYPTO_ALGAPI=y
CONFIG_CRYPTO_AEAD=y
CONFIG_CRYPTO_BLKCIPHER=y
CONFIG_CRYPTO_HASH=y
CONFIG_CRYPTO_RNG=y

#
# Crypto algorithms (safe to build as modules)
#
CONFIG_CRYPTO_CHACHA20=m
CONFIG_CRYPTO_POLY1305=m
CONFIG_CRYPTO_CURVE25519=m
CONFIG_CRYPTO_LIB_CHACHA20POLY1305=m
CONFIG_CRYPTO_LIB_CURVE25519=m

#
# WireGuard VPN
#
CONFIG_WIREGUARD=m
CONFIG_WIREGUARD_DEBUG=n

#
# OpenVPN (tun/tap devices)
#
CONFIG_TUN=m

#
# Netfilter core
#
CONFIG_NETFILTER=y
CONFIG_NETFILTER_ADVANCED=y
CONFIG_NETFILTER_XTABLES=m

#
# Connection tracking
#
CONFIG_NF_CONNTRACK=m
CONFIG_NF_CONNTRACK_IPV4=m
CONFIG_NF_CONNTRACK_IPV6=m

#
# NAT support
#
CONFIG_NF_NAT=m
CONFIG_NF_NAT_IPV4=m
CONFIG_NF_NAT_IPV6=m
CONFIG_NETFILTER_XT_TARGET_MASQUERADE=m
CONFIG_NETFILTER_XT_TARGET_NAT=m

#
# iptables compatibility
#
CONFIG_IP_NF_IPTABLES=m
CONFIG_IP_NF_FILTER=m
CONFIG_IP_NF_NAT=m
CONFIG_IP6_NF_IPTABLES=m
CONFIG_IP6_NF_FILTER=m
CONFIG_IP6_NF_NAT=m
CONFIG_NETFILTER_XT_MATCH_CONNTRACK=m
CONFIG_NETFILTER_XT_MATCH_STATE=m
```

This way one could always install the required modules from the feed and set things up manually.

closed 2026-10-09 by atvcaptain